jcfain/TCodeESP32 #8: WIFI_REASON_AUTH_EXPIRE

bestenator2, 2022-11-12:
I did the installation using the installer (flash.bat -> discord version 0253b).

I'm trying to setup the WIFI from the web client (192.168.1.1). I can't do it work. I have tried with two diferents routers (2 differents 2.4GH WIFI. Reviewing the arduino serial monitor, the debug trace say:

INFO: Establishing connection to SION3
.WARNING: Disconnected from station, attempting reconnection
INFO: Reason: 0
INFO: WIFI_REASON_AUTH_EXPIRE
..WARNING: Disconnected from station, attempting reconnection
INFO: Reason: 0
INFO: WIFI_REASON_AUTH_EXPIRE
.WARNING: Disconnected from station, attempting reconnection
INFO: Reason: 0
INFO: Unknown reason 0
..WARNING: Disconnected from station, attempting reconnection
INFO: Reason: 0
.WARNING: Disconnected from station, attempting reconnection
INFO: Reason: 205
INFO: Unknown reason 205
..WARNING: Disconnected from station, attempting reconnection
INFO: Reason: 205
.ERROR: Wifi timed out connection to AP

Thank in advance.

bestenator2, 2022-11-12:
Setting the WIFI properties on my ASUS router to Legacy Mode the connection works. It's appear that could be something related to the authentication mode used in the code.

jcfain, 2022-12-03:
WIFI_REASON_AUTH_EXPIRE is a code built into the ESP32s operating system. 
[](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#auth-phase)

Says it happens if it has not received any packets back from the AP/router within a specific time. I don't know if I can do anything about thing in the code to fix it. Ill leave this open till I get some time to look into it.
Maybe try changing some stuff on the router?

CryptoNickSoft, 2023-12-12:
Greetings! I've been struggling with the DOIT ESP32 DEVKIT V1 firmware for several days
Everything is stitched perfectly. I can connect to the device’s access daughter and make network settings.
The second blue LED on the device lights up.
I reboot the device after setup, the blue LED no longer lights up.
There should be a Wi-Fi connection, but I can't see it. The router does not provide such information.
I tried all available IP addresses, but there was no connection.

Your certificate contains gaps and a lot of things are not very clear.
The blue LED should light up after saving the network settings???
Will the ability to receive movements via usb com work or is it only necessary via wi-fi?
I couldn't set up Wi-Fi, but I have a USB connection. Can I use the device via USB? It's not working at the moment and I don't understand why. The servos are in a rigid state and do not move, although one periodically twitches when idle.
How to find out the IP of the board in Windows?
I ask for help, I don’t speak English and it’s hard for me to understand the documentation. (but very interesting)
If I didn’t have an Arduino Nano and the old firmware, I would have thought that the device didn’t work at all and abandoned the assembly, but that’s not the case...
I'm waiting for your help.

CryptoNickSoft, 2023-12-14:
The firmware version TCode_ESP32_Release_v0.3b-bldc is a little better.
The blue LED does not light up at all, but it is possible to connect to the Wi-Fi network and access the address http://tcode.local/
True, it takes a long time to connect and it is advisable to turn off the VPN on your computer.
The servos are still not controlled... They twitch periodically when idle and no more
I ordered a display for myself. Which Arduino pins should I connect the signal wires to?
I cannot pay for access to materials, since we are under sanctions.
I want to build a robot just for myself, but there are a lot of difficulties, as it turns out.

friendlyFriend4000, 2023-12-18:
> The firmware version TCode_ESP32_Release_v0.3b-bldc is a little better. The blue LED does not light up at all, but it is possible to connect to the Wi-Fi network and access the address [...]

There is a known issue with the USBC version of generic esp32. tcode.local won't load when the esp32 is powered by USB, try powering it via the VIN and GND pins. On the webui you will be able to set the pin for the display.

CryptoNickSoft, 2023-12-18:
I have a micro USB version. I was able to connect, I see the interface, but the servos are dead.
I'll try the new firmware version.
The display has arrived, which legs should I connect the data legs to?

CryptoNickSoft, 2023-12-18:
Thank you for your reply! Tell me which board legs to connect the display
screen and signal wires to.

Благодарю за ответ! Подскажите, к каким ножкам платы, подключить экран
дисплея, сигнальные провода.

friendlyFriend4000, 2023-12-19:
> I have a micro USB version. I was able to connect, I see the interface, but the servos are dead. I'll try the new firmware version. The display has arrived, which legs should I connect the data legs to?

It looks like the display is driven with I2C, which means you need to connect them to your SDA and SCL pins. Usually SDA is connected to GPIO 21 and SCL to GPIO 22. Additionally to that you need to enable the display in the webui.

CryptoNickSoft, 2023-12-19:
Thank you! The display worked but only for TCode_ESP32_Release_v0.3b-bldc

friendlyFriend4000, 2023-12-19:
> Thank you! The display worked but only for TCode_ESP32_Release_v0.3b-bldc

I think you have the wrong file. BLDC is for the ssr1 with the brushless dc motor.

CryptoNickSoft, 2023-12-19:
Yes, I understand this, but other firmwares do not work at all.

friendlyFriend4000, 2023-12-19:
> Yes, I understand this, but other firmwares do not work at all.

Can you describe in detail how you have your build setup.

CryptoNickSoft, 2023-12-19:
I'm flashing it, trying to access the web interface, but it doesn't work. The display does not light up.
The Wi-Fi password is correct.
It only worked on one firmware mentioned earlier. But servos do not work in more than one firmware.

friendlyFriend4000, 2023-12-19:
> I'm flashing it, trying to access the web interface, but it doesn't work. The display does not light up. The Wi-Fi password is correct. It only worked on one firmware mentioned earlier. But servos do not work in more than one firmware.

you need to describe in detail how you have everything hooked up. cables and pin numbers, everything. without such information i cannot help you

jcfain, 2024-01-29:
> The firmware version TCode_ESP32_Release_v0.3b-bldc is a little better. The blue LED does not light up at all, but it is possible to connect to the Wi-Fi network and access the address [...]

To start, dont put many thoughts into the blue LED. Thats not something that was planned and was just a by product of having a servo on pin 2 I believe. If your servos are not working then the firmware is either not flashed correct or theres something else wrong. The servos should work whether you have access to http://tcode.local or not.

Im closing this thread as its gotten off topic. I would suggest hopping on Tempests discord for help. Lots of people that can help there. Im usually helping there as well.

jcfain, 2024-01-29:
I came here to say I killed my long time friend (my first ESP32) a couple days ago and was forced to drop in a replacement and I am getting this error WIFI_REASON_AUTH_EXPIRE now too. I think this is due to a cheap cp2102 clone from a shady devkit manufacturer.
If I can fix it I will but otherwise im going to chop it up to faulty hardware. 
Thanks!

closed 2024-01-29 by jcfain

friendlyFriend4000, 2024-01-30:
I'd like to add that it mostly happens to all my USBC boards which have the CH340C chip