jcfain/TCodeESP32 #13: ESP32 DEVKIT V1 Bad Firmware

CryptoNickSoft, 2023-12-19:
ESP32 DEVKIT V1 ESP32-D0WD-V3 Rev 3 micrUSB version
Works partially or not at all.
I achieved partial work only with the firmware TCode_ESP32_Release_v0.3b-bldc
The display shows information about the access point and there is a connection.
I can connect it to the player, but the servos don't work.
With other firmware versions TCode_ESP32_Release_v0.3b и TCode_ESP32_Release_v0.34b-bldc
TCode_ESP32_Release_v0.34b does not work at all, the display does not turn on, there is no Wi-Fi connection, the servos do not work, there is communication via the COM port:

ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4

INFO main: ESP32 Chip model = ESP32-D0WD-V3 Rev 3
INFO main: This chip has 2 cores
INFO main: Chip ID: 5298060
INFO settings-handler: Read Settings file: /userSettings.json
INFO settings-handler: Read password file
INFO settings-handler: Load settings
INFO settings-handler: Last reset reason: Brownout reset (software or hardware)
INFO main: Version: 0.34b
INFO display-handler: Setting up display
INFO settings-handler: Scanning for I2C...
INFO settings-handler: I2C device found at address 0x3c, byte 60
INFO settings-handler: No I2C devices found in system, trying again...
INFO display-handler: No I2C devices found in system
INFO main: Starting in APMode

Brownout detector was triggered

Regarding the utility: flash.exe
If you click update, an error appears, saying the System cannot find the specified path
It is not intuitively clear at what stage you need to enter the Wi-Fi password
There is no choice of turning on the display with an indication of its settings, and this is important, because it is not clear what state the board is in.
There is no stopping logging to read errors and understand what is wrong. You have to turn off the device so that you can read the text.

Fortunately, the display should turn on automatically and show the state of the board.

CryptoNickSoft, 2024-01-05:
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4

INFO main: ESP32 Chip model = ESP32-D0WD-V3 Rev 3
INFO main: This chip has 2 cores
INFO main: Chip ID: 5298060
INFO settings-handler: Loading wifi info
INFO settings-handler: Loading json file /wifiInfo.json
INFO settings-handler: Loading common settings
INFO settings-handler: Loading json file /userSettings.json
INFO settings-handler: Last reset reason: Brownout reset (software or hardware)
INFO settings-handler: Loading motion profiles
INFO settings-handler: Loading json file /motionProfiles.json
INFO main: Version: 0.35b
INFO display-handler: Setting up display
INFO settings-handler: Scanning for I2C...
INFO settings-handler: I2C device found at address 0x3c, byte 60
INFO settings-handler: No I2C devices found in system, trying again...
INFO display-handler: No I2C devices found in system
INFO main: Starting in APMode

Brownout detector was triggered

CryptoNickSoft, 2024-01-05:
Absolutely nothing works! The display does not show information!
The board is connected via USB, there are no servos, the display is connected (it worked in one of the firmwares)
NO voltage dips! The computer is new, the cable is connected to the motherboard.

CryptoNickSoft, 2024-01-19:
I found what's wrong!
USB power problem “Brownout detector was triggered”
Namely, a lack of current due to the cord or computer, and not even every power adapter could cope with this.
Can’t even connect to Wi-Fi, although he tries
The ESP32 simply goes into defense and does not perform any actions!!!!

I'm more than sure that many people will have this problem.
Ways to solve it, as I see it. I need to add a display switching point to my firmware program so that I can see if the board starts up and it is advisable to display a message saying Low power
Although I don’t know how you power the board, because the connection is via wi-fi and a usb cable is not required...
But feeding it in the same circuit with servos is very bad, since they cause interference on the board. I'm not sure whether filters will save you.

Low power control can be disabled in the firmware code:
#include "soc/soc.h"
#include "soc/rtc_cntl_reg.h"

// in setup()
  WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0);

https://www.youtube.com/watch?v=_SIfywxHqL0

Make it possible to stop logging in the flasher window, because it is difficult to read the error code.

CryptoNickSoft, 2024-01-19:
https://stackoverflow.com/questions/73200166/exception-handling-for-brownout-detector-was-trigerred

CryptoNickSoft, 2024-01-19:
I found what's wrong!
USB power problem “Brownout detector was triggered”
Namely, a lack of current due to the cord or computer, and not even every power adapter could cope with this.
I discovered that there is no diode on my board and what I feed to VIN goes straight via USB to the computer.
Which is dangerous. I found one solution, unsolder the 0 ohm resistor on the ESR32 board and power it only from an external source, even during firmware. The issue of power filtering also arises, because there is a lot of interference from servos and this can negatively affect the operation of the device as a whole. Electrolyte and ceramics in the circuit, I doubt that they will help much... Separate power supplies are inconvenient...

jcfain, 2024-02-11:
> ESP32 DEVKIT V1 ESP32-D0WD-V3 Rev 3 micrUSB version Works partially or not at all. I achieved partial work only with the firmware TCode_ESP32_Release_v0.3b-bldc The display shows information about [...]

Thanks for the UX tips. I will attempt to make it more clear somehow.
As far as the other issue, these seem like hardware issues and not really issues with the firmware so I'm closing this request

closed 2024-02-11 by jcfain