jcfain/TCodeESP32 #16: New firmware

CryptoNickSoft, 2024-02-12:
Thank you for developing and improving the firmware. Unlike the author of the device...
As a beginner in constructing such robots, I saw many incomprehensible moments and described them in the discord.
But the author blocked my access. I don’t know why, either I didn’t like the activity, or my online translator translated from Russian too roughly...
In any case, I believe that you have read my posts and slightly modified the firmware.
Manual control is a must. Since many scripts are completely boring and only repeat movements on the screen, without giving sensations like in reality. It's good that you implemented this.
I believe the red button on the side should work like Break\Start
Moreover, Break is a sharp rise of the fleshlight upward and cancellation of Tcode reception
And Start is a smooth lowering of the fleshlight down and activation of Tcode reception (smooth, that’s the key word)
I hope why this is needed is clear. For example, when the power to the servos is turned off, one of them constantly jams if you try to move it manually, but it works normally when power is applied. It turns out that you can’t get out from under the device...
Vesa mounts would save the day here, but I haven’t been able to find the optimal one yet, although I’ve printed a lot of them.

CryptoNickSoft, 2024-02-12:
The new version of flash.exe does not find some libraries in the system and does not start, error messages appear

CryptoNickSoft, 2024-02-12:
I didn’t waste time searching for libraries, I flashed with the old version of flash.exe
The display worked immediately, setup was not difficult.
The firmware keeps getting better!

CryptoNickSoft, 2024-02-12:
libgcc_s_seh-1.dll
libwinpthread-1.dll
libstdc++-6.dll
Not found

I found and put these files in the program folder, but now there is still an error, without indicating the missing libraries.
I really don’t understand why such simple utilities should be written using platform-dependent languages.
On the same PureBasic, the program would weigh 100kb and would run on any kettle.

jcfain, 2025-01-19:
Closing due to duplicate post https://github.com/jcfain/TCodeESP32/issues/17

closed 2025-01-19 by jcfain